top-maker/modpoll #3: Add support for Daniel/Enron single register 32-bit mode

ruimarinho, 2022-03-28:
Hi,

I'm incredibly excited about the potential of this project given the many limitations of pymodbus and its integration in the Home Assistant project. The MQTT integration makes it a great candidate for IoT projects.

I was testing your library and I've noticed it also lacks support for the so called "Enron Modbus", or 32-bit values as one register instead of two. This is a particularly common modbus mode in Portuguese Smart Meters. It has been implemented in other python libs like https://github.com/u9n/enron-modbus. 

Is this something you could consider implementing? 

Thanks!

gavinying, 2022-03-29:
Hi @ruimarinho , thanks for your interest. Since this project is based on pymodbus and 32bit register support must be implemented in pymodbus level. However, I noticed there has already been some discussions around [#669](https://github.com/riptideio/pymodbus/issues/669), [#627](https://github.com/riptideio/pymodbus/issues/627). I would follow up once pymodbus supports it.

gavinying, 2023-07-16:
I will close it since no update for a while. Welcome to re-open if needed.

closed 2023-07-16 by gavinying