top-maker/modpoll #5: Add Option for Remove Local Echo

buddy-78, 2022-12-31:
Hi,
thank you for this great project.
I found out today that my converter generates a local echo and it can't be turned off. 
So my question is if an option can be added to remove local echoes.
Thanks!

gavinying, 2023-01-14:
Hi, would you be able to share more infomation? Because "local echo" is usually an option provided by a serial terminal tool, however `modpoll` doesn't have "local echo" feature builtin.

buddy-78, 2023-01-14:
Hi, 
i am using modpoll in modbus RTU. 
since my new usb-rs485 converter (FTDI FT232BL Chipset) echoes every sent byte back to its TX, modpoll gets checksum errors 
Some modbus master simulators like in my screenshot do removing local echo.
[image]
[image]

gavinying, 2023-03-01:
Close it due to no activity for a while.

closed 2023-03-01 by gavinying