top-maker/modpoll #8: Output should be rounded with multiplication factor

cfreire, 2023-01-13:
On using mopoll with Hager SM101C and apply factor the output should be rounded
Many thanks for you work. Very useful

__config csv__
```csv
device,sm101c,21,,,
poll,holding_register,50514,60,BE_BE

ref,U12,50514,uint32,r,V,0.01
ref,U23,50516,uint32,r,V,0.01
ref,U31,50518,uint32,r,V,0.01
ref,V1,50520,uint32,r,V,0.01
ref,V2,50522,uint32,r,V,0.01
ref,V3,50524,uint32,r,V,0.01

ref,Freq,50526,uint32,r,Hz,0.01
```

__output__
```txt
2023-01-13 15:56:14,905 | I | modpoll.modbus_task | Reading device:sm101c, FuncCode:3, Start_address:50514, Size:60... SUCCESS
===== references from device: sm101c =====
+----------------+------+---------+--------------------+
|      name      | unit | address |       value        |
+----------------+------+---------+--------------------+
|      U12       |  V   |  50514  |       237.35       |
|      U23       |  V   |  50516  |       410.38       |
|      U31       |  V   |  50518  |       236.18       |
|       V1       |  V   |  50520  |        0.0         |
|       V2       |  V   |  50522  | 237.73000000000002 |
|       V3       |  V   |  50524  | 236.64000000000001 |
|      Freq      |  Hz  |  50526  |        0.0         |
|       I1       |  A   |  50528  |        23.2        |
|       I2       |  A   |  50530  | 12.700000000000001 |
|       I3       |  A   |  50532  |        12.8        |
|       IN       |  A   |  50534  | 14.200000000000001 |

```

gavinying, 2023-01-13:
Yeah, it has just been fixed in the latest version 0.5.0. Please check. Thanks for your suggestion.

closed 2023-01-13 by gavinying

mr-manuel, 2023-02-27:
Hi,
I have more or less the same concern. In the log the value is rounded, but in the MQTT output not. I'm on the latest version `v0.5.1`.

```
publishing MQTT topic: modpoll/must-ph18-5548-plus, msg: {
    "charger-voltage-float": 54.400000000000006,
    "charger-voltage-absorption": 57.6,
    "battery-voltage-low": 34,
    "battery-voltage-high": 60,
    "charger-max-current": 80,
    "battery-ah": 200,
    "charger-work-state": 2,
    "charger-mptt-state": 1,
    "charger-charging-state": 1,
    "pv-voltage": 57.5,
    "battery-voltage-charger": 53.400000000000006,
    "charger-current": 2,
    "charger-power": 104,
    "temperature-radiator": 20,
    "temperature-external": 0,
    "relay-battery": 1,
    "relay-pv": 1,
    "error-charger": 0,
    "warning-charger": 0,
    "charger-rated-current": 80,
    "inverter-work-state": 3,
    "battery-voltage": 53.2,
    "inverter-voltage": 230.9,
    "grid-voltage": 230.8,
    "bus-voltage": 426.1,
    "control-current": 0,
    "inverter-current": 0,
    "grid-current": 1,
    "load-current": 0.9,
    "inverter-power": 0,
    "grid-power": -205,
    "load-power": 201,
    "load-percent": 3,
    "inverter-appearent": 0,
    "grid-appearent": 230,
    "load-appearent": 207,
    "inverter-react": 0,
    "grid-react": 106,
    "load-react": 45,
    "inverter-frequency": 0,
    "grid-frequency": 49.99,
    "temperature-ac-radiator": 17,
    "temperature-transformer": 44,
    "temperature-dc-radiator": 18,
    "relay-inverter": 1,
    "relay-grid": 1,
    "relay-load": 1,
    "relay-line": 0,
    "relay-dc": 1,
    "relay-eart": 0,
    "error-1": 0,
    "error-2": 0,
    "error-3": 0,
    "warning-1": 0,
    "warning-2": 0,
    "battery-power": -103,
    "battery-current": -1,
    "power-rated": 5500
}, qos: 0, RC: 0
```

gavinying, 2023-09-15:
Hi @mr-manuel , yes, this is on purpose, the screen printout is rounded for human eyes, but for MQTT output, IMHO, it is better to let client program handle rounding according to their specific resolution needs.