top-maker/modpoll #11: Consider add device 'IP ADDR' and 'port' to .csv file

cfreire, 2023-02-28:
I think it would a benefit  to add  `ip addr` and `port` to __device line__ in config file (.csv)

example: __sm101.csv__

```csv
device,QE2.2,21,192.168.50.68,502
poll,holding_register,50514,60,BE_BE

ref,U12,50514,uint32,r,V,0.01
ref,U23,50516,uint32,r,V,0.01
ref,U31,50518,uint32,r,V,0.01

ref,V1,50520,uint32,r,V,0.01
ref,V2,50522,uint32,r,V,0.01
ref,V3,50524,uint32,r,V,0.01
```
This will allow to simple call `$ modpoll -f sm101.csv`. The main objective is to record on the _.csv_ file the device origin for easy polling multiple devices.
Many thanks again
cf

gavinying, 2023-03-01:
@cfreire thanks for your interest. this is a good question in my mind as well. 

> Since we already have a config file, why not put all arguments into it? 

The reason I finally decide to keep the current format is, 
1. Try to make the modbus config file **reusable**, ideally it only describes the modbus specs from vendor;
2. Leave device-specific arguments in command line, so that it can be **dynamically** assigned by program or env, which is very useful in docker deployment;

Now you may ask why then `device_id` is still in the config file, that is because MODBUS RTU relies on host's serial port, which can only be **occupied by one process** at a moment for communication, basically we cannot run two `modpoll` commands on the same serial port, so it has to be defined in one modbus config file for all devices connected to one serial port. 

Hope this clarifies. 
Shaodong

cfreire, 2023-03-01:
Many thanks @gavinying for your kindly replay.

Well honestly, I'm writing an OPC UA Server that poll's about 100 devices, and uses your __modpool__ to initially verify the connection and values. Afterwards the `config.csv` file, goes into a poll directory with the rest 99 config files. I think it's easy to maintain 100 different config files, than a big one.
So for me it's better to add all connection data to the `config001.csv`, rather a separate into connection_location + poll_data

__I perfectly understand your reason's for not doing so and agree with them. Just different purposes!__

As Raymond Hettinger once said in a Python conference "We never know what will be the future utilization for our software". He claims to have written a flight simulator in AWK.
Again, many thanks for your work and contribution.

gavinying, 2023-03-01:
Glad to see `modpoll` tool helps in such a different way! Always welcome forking for your own needs, happy hacking!

closed 2023-03-01 by gavinying