top-maker/modpoll #16: writing registers with modpoll - how to do....?  ...reading works fine !

pythonlubo, 2023-06-21:
Hello together,
i use modpoll many time to read chinese inverters. ...works fine....
Next step - to give parameters to the inverters.
i understand csv-file....but i receive error messages from modpoll....
"UnboundLocalError: local variable 'ref' referenced before assignment"
The csv is:

device,wr1,1,,
poll holding_register, 300, 50, BE_BE
ref,cmax,332,Uint16,w

...it should write in Register 332 what on mqtt-broker is.... !?

someone have a idea....?
Thanks......

gavinying, 2023-07-16:
Hi, the holding_register writing function can be achieved via MQTT publish, take the online `modsim` as an example, when you run `modpoll` with MQTT broker specified, 
```
modpoll --tcp modsim.topmaker.net --config https://raw.githubusercontent.com/gavinying/modpoll/master/examples/modsim.csv --mqtt-host mqtt.eclipseprojects.io
```
the tool will automatically subscribe to the topic `modpoll/modsim001/set`, so if you want to write to a specific holding_register, you can use any MQTT client tool, like mosquitto_client or MQTTX, publish the following message to the target device's topic, e.g. `modpoll/modsim001/set` for device `modsim001`, 
```
{
  "object_type": "holding_register",
  "address": 40001,
  "value": 12
}
```
After the message succesfully published, you will see the value changed in `modpoll` output. 
[image]

Hope it helps. 
Thanks for your interest for modpoll. Cheers. 

Gavin

gavinying, 2023-07-16:
BTW, you can find more details [here](https://github.com/gavinying/modpoll#write-registers-via-mqtt-publish) in README.

pythonlubo, 2023-07-24:
Hi Gavin, with your documentation i've no positive result on my RPi-system....

now I've check what journalctl says to published messages on the modpoll RPi.....

My Inverter has a Remote Shutdown Register : {"object_type": "holding_register", "address": 420, "value": 0} to swich off (..and on)

Modpoll has in .csv the lines:

__________________________________________

device,wr1,1,,

poll,holding_register,200,50,BE_BE

ref,rs1,420,int16,rw       # ( rs1 means remote switch 1 )

_________________________________________

In mosquitto broker will publish the register in /hs36/wr/wr1/rs1/"1" (...for ON)

...works fine....

Now I published with mosquitto_pub:

mosquitto_pub -h xxx.xxx.xxx.xxx -p 1883 -t /hs36/wr/wr1/set -m "{"object_type": "holding_register", "address": 420, "value": 0}" for shut down the inverter...

journalctl shows an error:

Fail to parse json message: b'{object_type: holding_register, address: 420, value: 0}'

______________________________________________________________________________________

If I give the message without -m "{.........}"  as {....}

Error Message from mosquitto:

Error: Unknown option 'holding_register,'.

_______________________________________________

If I give the message with -m '{....}'

It will break the modpoll completly....

AttributeError: 'ModbusIOExeption' object has no attribute 'function_code'

_________________________________________________

...what other solutions I've to test....?

Thanks for help....

pythonlubo, 2023-07-24:
Hello again,
the right mosquitto_pub line is:

mosquitto_pub -h xxx.xxx.xxx.xxx -p 1883  -t /hs36/wr/wr1/set -m '{"object_type": "holding_register", "address": 406, "value": 1}'

(for example)

In journalctl i've see that the message arrive....after polling all csv-lines modpoll give the message:

 - Writing Register(s): device=wr1, address=406, value=1 -

...after that modpoll stops with errorcode:

return.result.function_code < 0x80
AttributError: 'ModbusIOException' object has no attribute 'function_code'

My Inverter need to write a register the functioncode 10H .....

modpoll has the possibility to give the function_code "10H" to the inverter ?

Thanks a lot.....

gavinying, 2023-07-27:
Yes, modpoll does support "10H", which writes mutiple registers, but in your case, looks only writing a single register, which shall use function code "06". 

In fact, modpoll supports the following function codes, fyi. 
- read coils (01)
- read descrete inputs (02)
- read holding registers (03)
- read input registers (04)
- write single coil (05)
- write single register (06)
- write multiple coils (15 or 0x0F)
- write multiple registers (16 or 0x10)

Can you share the complete logs? including the command you used to start modpoll for further investigation.

pythonlubo, 2023-07-27:
...now I try "minimalmodbus" to write the Inverter registers - it works fine....

The correct "minimalmodbus"-Line here is: 

instrument.write_register(324, 560, 0, 16)

324 is the registeraddress
560 is the value without decimal - means 56.0 Volt finaly
0 - i dont know for what....
16 is the right command to write the register

I think modpoll should use the "16" command to write - not "10" or "6"....
Ii it possible to advise modpoll to use the 16 command for writing registers...?
tks.....

gavinying, 2024-01-19:
Close it due to no activities for long time.

closed 2024-01-19 by gavinying