top-maker/modpoll #28: Exceptions when calling local config file

digispiders, 2024-01-26:
On fresh install of Raspbian Bookworm and in a fresh venv, calling modpoll results in the following traceback:
```
$modpoll --tcp 192.168.1.50 --config $PWD/modfelix.csv

modpoll - A New Command Line Tool for Modbus

2024-01-26 12:20:24,938 | I | modpoll.main | No MQTT host specified, skip MQTT setup.
2024-01-26 12:20:24,938 | I | modpoll.modbus_task | Loading config from: /home/pi/modfelix.csv
Traceback (most recent call last):
  File "/home/pi/modpollviron/lib/python3.11/site-packages/modpoll/modbus_task.py", line 377, in load_config
    response = s.get(file)
               ^^^^^^^^^^^
  File "/home/pi/modpollviron/lib/python3.11/site-packages/requests/sessions.py", line 602, in get
    return self.request("GET", url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/modpollviron/lib/python3.11/site-packages/requests/sessions.py", line 575, in request
    prep = self.prepare_request(req)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/modpollviron/lib/python3.11/site-packages/requests/sessions.py", line 486, in prepare_request
    p.prepare(
  File "/home/pi/modpollviron/lib/python3.11/site-packages/requests/models.py", line 368, in prepare
    self.prepare_url(url, params)
  File "/home/pi/modpollviron/lib/python3.11/site-packages/requests/models.py", line 439, in prepare_url
    raise MissingSchema(
requests.exceptions.MissingSchema: Invalid URL '/home/pi/modfelix.csv': No scheme supplied. Perhaps you meant https:///home/pi/modfelix.csv?

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/modpollviron/bin/modpoll", line 8, in <module>
    sys.exit(app())
             ^^^^^
  File "/home/pi/modpollviron/lib/python3.11/site-packages/modpoll/main.py", line 63, in app
    if not modbus_setup(args, event_exit):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/modpollviron/lib/python3.11/site-packages/modpoll/modbus_task.py", line 398, in modbus_setup
    load_config(args.config)
  File "/home/pi/modpollviron/lib/python3.11/site-packages/modpoll/modbus_task.py", line 385, in load_config
    parse_config(csv_reader)
  File "/home/pi/modpollviron/lib/python3.11/site-packages/modpoll/modbus_task.py", line 291, in parse_config
    device_name = row[1]
                  ~~~^^^
IndexError: list index out of range

```

gavinying, 2024-01-27:
Hi @digispiders, It looks like the config file parsing is failed. Could you share your config file?

digispiders, 2024-01-27:
[modfelix.csv](https://github.com/gavinying/modpoll/files/14070723/modfelix.csv)
File attached.

gavinying, 2024-01-27:
Please remove all comments in your config file, try [this](https://github.com/gavinying/modpoll/files/14070805/modfelix_mod.csv)

gavinying, 2024-01-27:
added a reminder in [example.csv](  https://github.com/gavinying/modpoll/blob/7d60884079bc59a95ebb05d0695e45412cf11c67/examples/example.csv#L41), also improved error message output in new version. 
Feel free to upgrade to v0.6.4 by `pip install -U modpoll`. 
Let me know if the issue has been solved on your end, thanks for reporting.

closed 2024-02-03 by gavinying