top-maker/modpoll #29: Issue specifying <object_type>+<start_address>

digispiders, 2024-01-29:
Unless Im missing something, here is whats happening:
When preparing the config file, it appears that its pulling an offset in the modbus addresses from the object type, and adding the starting_address value to it.  So, if you specify:
```
poll,holding_register,40001,10,BE_BE
ref,objectname,40001,uint16,r
```
It throws an error.  It appears that it might be adding 40001 from the type=holding_register to the specified address (40001) to be polling at the wrong place.
If you specify:
```
poll,holding_register,1,10,BE_BE
ref,objectname,1,uint16,r
```
It pulls from 40001 (the first holding_register).

Problem:
This conflicts with the documentation: [helloysd](https://helloysd.gitlab.io/modpoll/configure.html)

Suggested Resolution:
Should either pull offset from object_type and be documented as such, or ignore type and only pull from specified start_address.
[image]
[image]
[image]
[image]

Uploading config files and screen shots to demonstrate.

gavinying, 2024-01-30:
@digispiders Firstly, congratulate you have actually retrieved data according to your last screenshot. 

As you mentioned, instead of polling start from the standard holding_register address `40001`, 
```
poll,holding_register,40001,10,BE_BE
ref,objectname,40001,uint16,r
```
you end up with polling start from address `1`,
```
poll,holding_register,1,10,BE_BE
ref,objectname,1,uint16,r
```
This happens quite often because Modbus is an open protocol, different vendors have their own interpretation on the standard, that's why we have many different variants, e.g. zero-based vs one-based, big-endian vs small-endian, etc. Usually vendors will provide documents/examples to help developers for integration. 

In your case, your device vendor is using relative address (offset address) rather than absolute address for data comm. 

Hope that helps for clarification.

digispiders, 2024-01-30:
Okay.  It was a little unclear to me based on the documentation, and probably also due to my poor understanding of modbus.  I assumed that an address was an address, which was probably wrong to assume.  Feel to close.

closed 2024-02-03 by gavinying