top-maker/modpoll #30: Modbus TCP RTU Framer

nate8199, 2024-02-10:
I'm having issues reading a serial device that is plugged into a TCP Serial Server with Modpoll
I can get it to work with plain pymodbus in VSCode with the attached code, not sure if I'm missing an option, doing something wrong, or if this type of device doesn't work by default.

#Python

```
import time
import pymodbus
from pymodbus.client import ModbusBaseClient, ModbusTcpClient, ModbusSerialClient
from pymodbus.framer import ModbusFramer, ModbusRtuFramer

client = ModbusTcpClient('192.168.0.63', port=502, framer=ModbusRtuFramer)

client.connect()
print(f'Connected\n')
while True:
    data = client.read_input_registers(address=0, count=10, slave=40)
    
    voltage = data.registers[0] / 10.0 # [V]
    current = (data.registers[1] + (data.registers[2] << 16)) / 1000.0 # [A]
    power = (data.registers[3] + (data.registers[4] << 16)) / 10.0 # [W]
    energy = data.registers[5] + (data.registers[6] << 16) # [Wh]
    frequency = data.registers[7] / 10.0 # [Hz]
    powerFactor = data.registers[8] / 100.0
    alarm = data.registers[9]

    print('Voltage: ', voltage)
    print('Current: ', current)
    print('Power: ', power) # active power (V * I * power factor)
    print('Energy: ', energy)
    print('Frequency: ', frequency)
    print('Power factor: ', powerFactor)
    print(f'Alarm: {alarm}\n')
    time.sleep(1)
```

#CSV (I have tried both Input and Holding)

```
device,desk_ac_current,40,,
poll,input_register,0,10,BE_BE
ref,input_reg00,0,int16,r,V,.1
```

#Command
`modpoll --tcp 192.168.0.63 --config desk_ac_current.csv
`

gavinying, 2024-02-11:
Hi @nate8199, thanks for your interest. You are right. The current version of `modpoll` is using SOCKET framer for tcp client and RTU framer for rtu client by default. It doesn't support your use case, which requires RTU framer but over tcp client. Thanks to your report, I understand such use cases might be useful when people tend to use RTU device over TCP network, I am going to add this feature in the next version of `modpoll`. Stay tuned.

gavinying, 2024-02-12:
Released a new version `0.7.0` with a newly added argument `--framer` and a few other improvements. 

@nate8199 Please help to test with your setup and let me know the outcome, since it is not easy to replicate such a unique setup on my side. 

The following command for your reference, feel free to adjust,
```
modpoll --tcp 192.168.0.63 --config desk_ac_current.csv --framer rtu
```
Possible choices of framer are: [ascii, binary, rtu, socket]. Refer to [here](https://helloysd.gitlab.io/modpoll/usage.html#Named%20Arguments) for argument details.

nate8199, 2024-02-16:
Thank you @gavinying, this works now with the new framer option

```
nate@framework:~/Code> modpoll --tcp 192.168.0.63 --tcp-port 502 --framer rtu --config desk_ac_current.csv

modpoll - A New Command Line Tool for Modbus

2024-02-16 16:00:28,888 | I | modpoll.main | No MQTT host specified, skip MQTT setup.
2024-02-16 16:00:28,888 | I | modpoll.modbus_task | Loading config from: desk_ac_current.csv
2024-02-16 16:00:28,889 | I | modpoll.modbus_task | Adding new device desk_ac_current
2024-02-16 16:00:28,889 | I | modpoll.modbus_task | Add poller (start_address=0, size=10) to device desk_ac_current
2024-02-16 16:00:28,889 | I | modpoll.modbus_task | Polling 1 device(s)...
2024-02-16 16:00:28,889 | I | modpoll.main |  ====== modpoll polling at rate:10.0s, actual:10.0s ======
2024-02-16 16:00:28,974 | I | modpoll.modbus_task | Reading device:desk_ac_current, FuncCode:4, Start_address:0, Size:10... SUCCESS
===== references from device: desk_ac_current =====
+-------------+-------+---------+-------+
|     name    |  unit | address | value |
+-------------+-------+---------+-------+
|   voltage   |   V   |    0    |  122  |
|     amps    |   A   |    1    | 0.073 |
|    power    |   W   |    3    |  4.2  |
|    energy   |   Wh  |    5    |  2227 |
|  frequency  |   Hz  |    7    |   60  |
| powerfactor |   pf  |    8    |  0.47 |
|    alarm    | alarm |    9    |   0   |
+-------------+-------+---------+-------+
Done.
```