top-maker/modpoll #31: Ommiting Prior Addresses of a Specified Coil

tteddy623, 2024-02-15:
Config file: 

device,relay1,1
poll,discrete_input,4,1,LE_LE
ref,relay1,4,bool8,r

Expected Output:

+--------+------+---------+---------------------------------------------------------+
|  name  | unit | address |                          value                          |
+--------+------+---------+---------------------------------------------------------+
| relay1 | None |    4    |                            [False] |
+--------+------+---------+---------------------------------------------------------+

Observed Output:

+--------+------+---------+---------------------------------------------------------+
|  name  | unit | address |                          value                          |
+--------+------+---------+---------------------------------------------------------+
| relay1 | None |    4    | [False, False, True, False, False, False, False, False] |
+--------+------+---------+---------------------------------------------------------+

Any change to the config either discrete_input or coil, etc results in a [NONE] value if I can manage to get it to filter out the rest of the addresses. The polling is working fine its just that there should be a way to obtain a single coil's state.  The output above makes for MQTT messaging impossible as other relay states will ruin the succcess parameters.

gavinying, 2024-02-16:
@tteddy623 The output format is actually following the Modbus standard, 

[image]

As above mentioned, Modbus `read coils` response is alwasys 1 or n bytes, `modpoll` tool will then forward the complete data to MQTT, but as data consumer, you can always select the target bit(s) from the received bytes on application level.

gavinying, 2024-04-04:
Close it since no activity for long time.

closed 2024-04-04 by gavinying