top-maker/modpoll #38: Added logic to only push mqtt messages when a change is outside of a range

mitchell-pioneer, 2024-06-18:
I added another change whereby mqtt messages are only sent when the value is outside of a range of values.  Currently changes are sent if the two values are not equal.  In practice, if the value is .0000001 different it will still send it.   I added an upper or lower change by a percentage. 

I made these src changes 

-CSV file has a new column
-New command line switch
-New logic in modbus_task::modbus_publish

I'm testing it now.  Let me know if this is something that you want to commit.

gavinying, 2024-06-19:
Yeah, that's a nice feature. Also your approach looks good to me. 
Here are some points for your consideration, 
1. Backwards compatibility: Current users don't have to modify their existing commands and config files for upgrade
2. The new code shall allow those additional columns in csv file to be empty if this feature is not required. 
3. Consider adding two columns (low bound / high bound) to make it more generic, which can be used for abnormal detection / alarm use case. Many device vendors provide such register value range in their Modbus documents. I believe it will also cover your use case. 

Your contribution is most welcome. Thanks.

mitchell-pioneer, 2024-06-19:
Great suggestions.  

I am also considering modifying the docker configuration.   I'm using Portainer, and it allows  you to specify environment variables from inside that tool.   I think I should add it in the contributions section as it will break the existing system.