top-maker/modpoll #57: Data transformation

dakhnod, 2024-09-11:
Thanks for your great work!

I am reading some registers that need a factor and an offset to be applied to the register value. Is there any way to achieve this through modpoll in the configuration?

gavinying, 2024-09-12:
Hi Daniel, current version only support applying a ratio but not offset. 

However, if you use MQTT as interface, I would recommend you to process data by subscribing on raw data topic and publish processed data to new topics. By doing this way, you always keep the first hand raw data and have the flexibility to apply different algorithms or even chain them as a pipeline.

njh, 2024-11-06:
[Node-RED](https://nodered.org/) is great for this!

mircsicz, 2025-01-06:
Here's how I've done simular on a Mbus-Meter with [NodeRed](
https://community.home-assistant.io/t/reading-energy-meter-via-wired-m-bus/591412/3?u=mircsicz)