top-maker/modpoll #69: Adding config file for SDM120M

njh, 2024-10-28:
Thanks for making modpoll - it looks like a great tool!

I have been trying to configure it for a Eastron SDM120M and looked at [SDM72D-M.csv](https://github.com/gavinying/modpoll/blob/master/contrib/eastron/SDM72D-M.csv) as an example.
They are very similar but I couldn't get the configuration in the example file to work.

Do you know if [SDM72D-M.csv](https://github.com/gavinying/modpoll/blob/master/contrib/eastron/SDM72D-M.csv) has ever been tested?

Although the documentation refers to Input Register `30001`. I think you actually have to read from `0` for the voltage:
[image]

This seems to be confirmed by this Python implementation:
https://github.com/nmakel/sdm_modbus/blob/master/src/sdm_modbus/sdm.py

Is it possible to specify addresses in hex? Or only decimal?

gavinying, 2024-10-28:
Hi @njh , thanks for your interest in modpoll tool. 

I have double checked the config and vendor's document, I think the current config file is only a direct translation of vendor's register map following some kind of Modbus conventions, however the vendor obviously implements it in a different way. 

I think the below updated config shall work with the meter,
```
device,eastron001,1,,,
poll,input_register,0,36,BE_BE,
ref,PHASE_1_LINE_TO_NEUTRAL_VOLTS,0,float32,r,V
ref,PHASE_2_LINE_TO_NEUTRAL_VOLTS,2,float32,r,V
ref,PHASE_3_LINE_TO_NEUTRAL_VOLTS,4,float32,r,V
ref,PHASE_1_CURRENT,6,float32,r,A
ref,PHASE_2_CURRENT,8,float32,r,A
ref,PHASE_3_CURRENT,10,float32,r,A
ref,PHASE_1_ACTIVE_POWER,12,float32,r,W
ref,PHASE_2_ACTIVE_POWER,14,float32,r,W
ref,PHASE_3_ACTIVE_POWER,16,float32,r,W
ref,PHASE_1_APPARENT_POWER,18,float32,r,VA
ref,PHASE_2_APPARENT_POWER,20,float32,r,VA
ref,PHASE_3_APPARENT_POWER,22,float32,r,VA
ref,PHASE_1_REACTIVE_POWER,24,float32,r,VAR
ref,PHASE_2_REACTIVE_POWER,26,float32,r,VAR
ref,PHASE_3_REACTIVE_POWER,28,float32,r,VAR
ref,PHASE_1_POWER_FACTOR,30,float32,r,
ref,PHASE_2_POWER_FACTOR,32,float32,r,
ref,PHASE_3_POWER_FACTOR,34,float32,r,
```
Please feel free to modify it for your actual needs. 
Welcome your PR once you confirm it on your end :) 

Regarding your suggestion to allow specify addresses in hex, I think it is definately a good one and will add it in the coming release. Stay tuned. Thanks.

njh, 2024-11-05:
Got it working 🙂 

Poll time is pretty long (24sec) - even with baud rate increased to 9600.
I will submit a PR for this to contrib.
I will delete the things I don't actually need on my setup.

```
Device: SDM120M
+------------------------------------+-----------+-----------+
| Reference                          |     Value | Unit      |
+------------------------------------+-----------+-----------+
| voltage                            |     244.1 | Volts     |
| current                            |     0.054 | Amps      |
| power_active                       |       5.9 | Watts     |
| power_apparent                     |      7.54 | VA        |
| power_reactive                     |       0.0 | VAr       |
| power_factor                       |       0.7 |           |
| phase_angle                        |       0.0 | Degrees   |
| frequency                          |      50.0 | Hz        |
| import_energy_active               |   342.284 | kWh       |
| export_energy_active               |       0.0 | kWh       |
| import_energy_reactive             |     0.054 | kWh       |
| export_energy_reactive             |   132.226 | kWh       |
| total_demand_power_active          |     5.128 | Watts     |
| maximum_total_demand_power_active  |  2736.674 | Watts     |
| import_demand_power_active         |     5.128 | Watts     |
| maximum_import_demand_power_active |  2736.674 | Watts     |
| export_demand_power_active         |       0.0 | Watts     |
| maximum_export_demand_power_active |     0.033 | Watts     |
| total_demand_current               |     0.054 | Amps      |
| maximum_total_demand_current       |    11.211 | Amps      |
| total_energy_active                |   342.284 | kWh       |
| total_energy_reactive              |    132.28 | kVArh     |
| demand_time                        |       1.0 | Seconds   |
| demand_period                      |      60.0 | Seconds   |
| relay_pulse_width                  |      60.0 | Millisecs |
| network_parity_stop                |       0.0 |           |
| meter_id                           |       1.0 |           |
| baud                               |       2.0 |           |
| pulse1_output_mode                 |       4.0 |           |
| display_scroll_timing              |         0 | Seconds   |
| pulse1_divisor                     |         0 |           |
| measurement_mode                   |         2 |           |
| indicator_mode                     |       0.0 |           |
| meter_code                         |        32 |           |
| software_version                   |       516 |           |
+------------------------------------+-----------+-----------+
2024-11-05 20:47:34,564 | I | modpoll.main |  === Modpoll is polling at rate:10.0s, actual:24.136957s ===
```

gavinying, 2024-11-06:
Great to hear it's working now :)
BTW, you can optimize the poll time by adjusting `--interval` argument (default: 1s), which set the interval between each poll for device to response the previous query), so for example, if you are sure your device can response within 200ms, then it's safe to set `--interval 0.2`.

njh, 2024-11-06:
Aha! I didn't realise that. I can actually set it to `--interval 0` and it works fine!

```
Device: SDM120M
+------------------------------------+-----------+-----------+
| Reference                          |     Value | Unit      |
+------------------------------------+-----------+-----------+
| voltage                            |     244.1 | Volts     |
| current                            |      0.19 | Amps      |
| power_active                       |       9.3 | Watts     |
| power_apparent                     |    43.114 | VA        |
| power_reactive                     |     -42.2 | VAr       |
| power_factor                       |     0.211 |           |
| phase_angle                        |       0.0 | Degrees   |
| frequency                          |      50.0 | Hz        |
| import_energy_active               |   355.601 | kWh       |
| export_energy_active               |       0.0 | kWh       |
| import_energy_reactive             |     0.057 | kWh       |
| export_energy_reactive             |   133.053 | kWh       |
| total_demand_power_active          |   541.338 | Watts     |
| maximum_total_demand_power_active  |  2736.674 | Watts     |
| import_demand_power_active         |   541.338 | Watts     |
| maximum_import_demand_power_active |  2736.674 | Watts     |
| export_demand_power_active         |       0.0 | Watts     |
| maximum_export_demand_power_active |     0.033 | Watts     |
| total_demand_current               |     2.287 | Amps      |
| maximum_total_demand_current       |    11.211 | Amps      |
| total_energy_active                |   355.601 | kWh       |
| total_energy_reactive              |    133.11 | kVArh     |
| demand_time                        |       1.0 | Seconds   |
| demand_period                      |      60.0 | Seconds   |
| relay_pulse_width                  |      60.0 | Millisecs |
| network_parity_stop                |       0.0 |           |
| meter_id                           |       1.0 |           |
| baud                               |       2.0 |           |
| pulse1_output_mode                 |       4.0 |           |
| display_scroll_timing              |         0 | Seconds   |
| pulse1_divisor                     |         0 |           |
| measurement_mode                   |         2 |           |
| indicator_mode                     |       0.0 |           |
| meter_code                         |        32 |           |
| software_version                   |       516 |           |
+------------------------------------+-----------+-----------+
2024-11-06 21:10:26,119 | I | modpoll.main |  === Modpoll is polling at rate:10.0s, actual:10.001245s ===
```

closed 2024-11-15 by gavinying