top-maker/modpoll #71: Multiple devices?

njh, 2024-11-06:
I have multiple [Heatmiser Modbus Room Thermostats](https://www.heatmiser.com/modbus-thermostat-series/).

Is it possible to describe a device type and then specify multiple device ids for that device type?
Or do you have to repeat the configuration for each device?

gavinying, 2024-11-07:
It is possible for Modbus TCP devices but not Modbus RTU devices.

njh, 2024-11-10:
How do you configure multiple Modbus TCP devices?

gavinying, 2024-11-15:
Since TCP device's IP address info is not included in the config file, so the config can be reused for all same type devices. To connect to multiple devices, I usually use multiple docker containers in a docker-compose file. 
For example, we can run multiple `modsim` devices locally or on cloud and connect to them via a single publicly shared `modsim` config file.