top-maker/modpoll #75: doc: enhance example by showing usage of "scale"

mi-hol, 2024-12-03:
Thank you so much with this great tool!

Digital power meters measure in factions of Watts but relevant for invoicing are kWh. 
Therefore the "scale" parameter is provided in configuration but unfortunately the current examples don't show its usage.

I guess adding "scale" would apply to Example 2: SCPM-S6 Power Meter.
but because I can't test it, providing a tested PR is not feasible for me.
Could you please take care of this proposal?

gavinying, 2024-12-12:
Hi, thanks for your interest in this tool. 

You are right that we provide "scale" parameter in Modpoll config file, the main purpose is to conform with vendor's Modbus protocol, as we know, some device vendors will apply a "scale" or "gain" on to their raw output so that they could use integer number to output float values. 
For example, HUAWEI_SUN2000_245KTL28KTL_MODBUS_Interface defines the PV voltages like below, 
[image]
there is a "gain" (same as "scale") value 10 applied to each PV voltage output, meaning if you got raw data 2405, the actual output is 2405/10 = 240.5 (V). 
And this is what Modpoll tool actually do via the config file defined, 
[image]

See Huawei smart logger example for details, 
https://github.com/gavinying/modpoll/tree/master/contrib/huawei/sun2000

Although it is not limited to use "scale" for your purpose, but the design principle is to output data according to vendor's specs and leave the data processing stage to the other user preferred programs/tools. 

Hope that clarifies.

mi-hol, 2024-12-12:
> See Huawei smart logger example for details, https://github.com/gavinying/modpoll/tree/master/contrib/huawei/sun2000

Thanks for providing a reference to an example, unfortunately this example is not mentioned in the documentation at https://gavinying.github.io/modpoll/configure.html#example-2-scpm-s6-power-meter-modbus-rtu-device.
Enhancing the existing documentation was the point I tried to suggest (i.e.  Example 2: SCPM-S6 Power Meter).
Does that make sense?

commit 8b4319c0 (2024-12-17): docs: Add reference scale usage in examples

closes #75
closed 2024-12-17 by gavinying