top-maker/modpoll #76: enhancement: display a better understandable error message

mi-hol, 2024-12-06:
Maybe a better understandable error message (including the string) could be displayed?

The reason for error is a wrong register address or length of string in configuration
```
modpoll\modbus_task.py", line 172, in _decode_and_update_reference
    decoder.decode_string(ref.ref_width * 2).decode("utf-8").rstrip("\x00")
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd7 in position 8: invalid continuation byte
```

gavinying, 2024-12-12:
thanks for your suggestion, will take it into consideration.

closed 2024-12-17 by gavinying

mi-hol, 2024-12-18:
Thanks for implementing an improvement!
When will the new version 1.4.1 become available for download?

mi-hol, 2025-01-05:
> When will the new version 1.4.1 become available for download?

@gavinying may I kindly ask what is holding up merging of PR #78  to get the fix released?

gavinying, 2025-01-06:
It will be released soon, just to avoid upgrading during holiday seasons:)

gavinying, 2025-01-06:
[v1.4.1](https://github.com/gavinying/modpoll/releases/tag/v1.4.1)  has been released. Feel free to download.

commit 505c76c3 (2024-12-17): chore: Improve error messages

Closes #76