top-maker/modpoll #77: bug: max string length displayed is 8

mi-hol, 2024-12-06:
Please increase max length of string to actually show the defined length

gavinying, 2024-12-12:
Could you elaborate a bit? 
1. Your setup
2. The expectation
3. Actual result
4. Your analysis or suggestion

mi-hol, 2024-12-12:
1. setup: config.csv
`device,FroniusGen24Sim,1,,
poll,holding_register,40004,32,BE_BE
ref,Manufacturer,40004,string16,r
ref,Model,40020,string16,r`

2. Actual result

[image]

3.  expected output:

[image]

 4. Your analysis or suggestion

a) width of "value" column is currently 11 characters but I would expect a width matching with defined maximum string length from config.cvs (16 in above example)

Does this explain the issue better? 
Happy to add more

gavinying, 2024-12-17:
I could understand your issue, but unfortunately I am not able to reproduce it on my end. 

By right, the prettytable shall auto adjust the column width according to the string value, see below example, 
[image]

Please double check your config file if the reference value polled is actually cut off, adjust your poll range and reference size if needed.

mi-hol, 2024-12-17:
> unfortunately I am not able to reproduce it on my end.
> ... prettytable shall auto adjust the column width

Thanks for investigating!
My setup is:
- Windows11
- PowerShell 7.4.x
- Python 3.12.8

I was unable to find the version of prettytable used in my setup due to below strange behavior.
Any hint how to find it?
What version of prettyprint should be installed?

```
PS C:\Users\U1\tests> pip freeze
argcomplete==3.5.1
click==8.1.7
colorama==0.4.6
packaging==24.2
pipx==1.7.1
platformdirs==4.3.6
pyModbusTCP==0.3.0
pyserial==3.5
userpath==1.9.2
PS C:\Users\U1\tests>  modpoll --tcp 192.168.0.178 --config FroniusGen24.debug.csv -1

Modpoll v1.4.0 - A New Command-line Tool for Modbus and MQTT

2024-12-17 14:07:30,492 | I | modpoll.main | No MQTT host specified, skip MQTT setup.
2024-12-17 14:07:30,492 | I | modpoll.modbus_task | Loading config from: FroniusGen24.debug.csv
2024-12-17 14:07:30,500 | I | modpoll.modbus_task | Added 1 device(s)...
2024-12-17 14:07:30,500 | I | modpoll.main | Loaded 1 Modbus config(s).
2024-12-17 14:07:30,500 | I | modpoll.main |  === Modpoll is polling at rate:10.0s, actual:10.0s ===

Device: FroniusGen24Sim
+--------------+----------+------+
| Reference    |    Value | Unit |
+--------------+----------+------+
| Manufacturer |  Fronius |      |
| Model        | Smart Me |      |
| Version      |     Fron |      |
| Leistung     |  -289.57 | Watt |
| Einspeisung  |   15.218 | kWh  |
| Verbrauch    |   251.16 | kWh  |
+--------------+----------+------+
PS C:\Users\U1\tests> python
Python 3.12.8 (tags/v3.12.8:2dc476b, Dec  3 2024, 19:30:04) [MSC v.1942 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from prettytable import PrettyTable
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'prettytable'
>>>

```

mi-hol, 2024-12-17:
Additional info likely related
```
pipx upgrade prettytable
Package is not installed. Expected to find C:\Users\U1\pipx\venvs\prettytable, but it does not exist.

PS C:\Users\U1> pipx install prettytable

No apps associated with package prettytable or its dependencies. If you are attempting to install a library, pipx
should not be used. Consider using pip or a similar tool instead.
PS C:\Users\U1> pip install prettytable
Collecting prettytable
  Using cached prettytable-3.12.0-py3-none-any.whl.metadata (30 kB)
Collecting wcwidth (from prettytable)
  Using cached wcwidth-0.2.13-py2.py3-none-any.whl.metadata (14 kB)
Using cached prettytable-3.12.0-py3-none-any.whl (31 kB)
Using cached wcwidth-0.2.13-py2.py3-none-any.whl (34 kB)
Installing collected packages: wcwidth, prettytable
Successfully installed prettytable-3.12.0 wcwidth-0.2.13
```

then started new shell but issue remains

```
PowerShell 7.4.6
PS C:\Users\U1> cd C:\Users\U1\Documents\tests
PS C:\Users\U1\Documents\tests>  modpoll --tcp 192.168.0.178 --config FroniusGen24.debug.csv -1

Modpoll v1.4.0 - A New Command-line Tool for Modbus and MQTT

2024-12-17 14:33:00,302 | I | modpoll.main | No MQTT host specified, skip MQTT setup.
2024-12-17 14:33:00,302 | I | modpoll.modbus_task | Loading config from: FroniusGen24.debug.csv
2024-12-17 14:33:00,302 | I | modpoll.modbus_task | Added 1 device(s)...
2024-12-17 14:33:00,302 | I | modpoll.main | Loaded 1 Modbus config(s).
2024-12-17 14:33:00,302 | I | modpoll.main |  === Modpoll is polling at rate:10.0s, actual:10.0s ===

Device: FroniusGen24Sim
+--------------+----------+------+
| Reference    |    Value | Unit |
+--------------+----------+------+
| Manufacturer |  Fronius |      |
| Model        | Smart Me |      |
| Version      |     Fron |      |
| Leistung     |  -111.14 | Watt |
| Einspeisung  |   15.292 | kWh  |
| Verbrauch    |   251.16 | kWh  |
+--------------+----------+------+

```

mi-hol, 2024-12-17:
I looked thru the tests quickly and it seems that the [code for handling "string" decodes starting at line 180](https://github.com/gavinying/modpoll/blob/8b4319c014be71bb5ac121888cc7dd9b8f552765/modpoll/modbus_task.py#L180) is not executed by any of the currectly defined tests.
Maybe a test case could be added for a 16 characters long string?

mi-hol, 2024-12-17:
> Maybe a test case using modsim.topmaker.net device could be added for a 16 characters long string?
Ideal for tests would be to extract:

```
            "VendorUrl": "https://github.com/riptideio/pymodbus/",
            "ProductName": "Pymodbus Server",
```

mi-hol, 2024-12-18:
I revised my configuration and get longer than 8 character strings.
But I still believe there is a bug in either documentation or implementation for string extracts.
Therefore adding a test case (as suggested in #79)  should be the first step for clarification

new config:
```
device,FroniusGen24Sim,1,,
poll,holding_register,40004,64,BE_BE
ref,Manufacturer,40004,string32,r
ref,Model,40020,string32,r
poll,holding_register,40004,80,BE_BE
ref,Manufacturer2,40004,string32,r
ref,Model2,40020,string32,r

poll,holding_register,40044,16,BE_BE
ref,Version,40044,string16,r
poll,holding_register,40071,30,BE_BE
ref,Leistung,40097,float32,r,Watt

poll,holding_register,40129,10,BE_BE
ref,Einspeisung,40129,float32,r,kWh,0.001
ref,Verbrauch,40137,float32,r,kWh,.001

```

new output:
```
Device: FroniusGen24Sim
+---------------+-----------------+------+
| Reference     |           Value | Unit |
+---------------+-----------------+------+
| Manufacturer  |         Fronius |      |
| Model         | Smart Meter 63A |      |
| Manufacturer2 |            None |      |
| Model2        |            None |      |
| Version       |         Fronius |      |
| Leistung      |          -18.77 | Watt |
| Einspeisung   |          15.599 | kWh  |
| Verbrauch     |         256.519 | kWh  |
+---------------+-----------------+------+
```

closed 2024-12-18 by mi-hol