top-maker/modpoll #79: Add a test case using modsim.topmaker.net device for string16

mi-hol, 2024-12-18:
> Maybe a test case using modsim.topmaker.net device could be added for a 16 characters long string?

Ideal for tests would be to extract:

```
            "VendorUrl": "https://github.com/riptideio/pymodbus/",
            "ProductName": "Pymodbus Server",
```

_Originally posted by @mi-hol in https://github.com/gavinying/modpoll/issues/77#issuecomment-2549020964_

mi-hol, 2024-12-18:
and field "MajorMinorRevision": "3.1.0"

gavinying, 2024-12-22:
In fact, the register's string length shall be according to device specification, it is not configured or decided by modpoll tool.

For your request, we need to modify modsim code to simulate a 16-character long string value, see https://github.com/gavinying/modsim/commit/392113d7586cc0d89fb40ef40ff142b116b30a79#diff-1214100fc30f036ad98e2a546fcb20b6fb7a7cf375bff7fcbb4a099590f6cbfdR97

Now, you could use below config to retrieve 16-byte long string value,
```
device,modsim,1,,
poll,holding_register,40036,8,BE_BE
ref,holding_reg19,40036,string16,rw
```
I will later update the modpoll example config files to include above usages as well. thanks. 

Btw, the modsim is only a very simple example of Modbus simulator (all logics are in one python file), you could fork it and easily make changes to adapt to your further tests.

gavinying, 2025-01-03:
Sorry for the late response, you are absolutely right, 16-character long string is 8 words, so here we shall poll 8 registers in the config file. 
Hope you have already get the problem solved.

gavinying, 2025-01-07:
I have to agree that the current document is far from covering all features or potential usecases of `modpoll` tool. the tool itself is quite flexible and can be used in many senarios, I'll continuously improve our documents. Thanks for your suggestions. 

Meanwhile, I would also like to say, I always **welcome contributions from users**, because people love to see and learn from actual examples, you might have just come through a hard time (due to non-standard vendor implementation or even incorrect vendor specs) to figure out a modpoll config for a specific device, now the other people happens to use the same vendor might be benefit from your work if you could share it. It could be a config file or any writeups of your experience or tips. **PULL REQUESTS ARE WELCOME**.

In fact, this is the gist of creating this `modpoll` tool, which made it different from the other traditional non-sharable `modpoll` tools. 

Happy coding and sharing :)

mi-hol, 2025-01-07:
> I always **welcome contributions from users**

I plan to contribute like I did in many other OSS projects already.
Unfortunately I've not mastered ModBus yet but modpoll is helping a lot.

I try to read data from devices supporting [SunSpec](https://sunspec.org/sunspec-technology-overview/) but have to rely currently on an ESP32-based simulator. This emulator is written in Tasmota script and suffers from a number of issues I hope to fix.
To complicated matters further, the emulator's source code is currently only available on a [file sharing website](https://pastebin.com/11zX4kG5) with [description on a different website](https://ottelo.jimdofree.com/stromz%C3%A4hler-auslesen-tasmota/) :(
I'm working with with the original author to publish on GH but with limited success yet.

the ModBus related code is below

current power in W = 40097
total Wh export = 40129 
total Wh Import = 40137 

`
switch reg
  case 768
        data={0}
 
    case 1706
        data={0}
 
    case 40000
        data={21365 28243}
 
    case 40002
        data={1 65}
 
    case 40004
        data={70 114 111 110 105 117 115 0 0 0 0 0 0 0 0 0}
        data={83 109 97 114 116 32 77 101 116 101 114 32 54 51 65 0}
        data={0 0 0 0 0 0 0 0}
        data={0 0 0 0 0 0 0 0}
        data={48 48 48 48 48 48 48 49 0 0 0 0 0 0 0 0}
        data={240 213}
 
    case 40069
        data={213 124}
 
    case 40071
        data={0 0 0 0 0}
        data={0 0 0 0 0}
        data={0 0 0 vsml1 0}
        data={0 0 0 0 0}
        data={0 0 0 0 0}
        data={0 0 0 0 0}
        dform=3
        nval=nval/2
 
    case 40129
        data={vsml3 0 0 0 vsml2}
        data={0 0 0 0 0}
        data={0 0 0 0 0}
        data={0}
        dform=3
        nval=nval/2
 
    case 40193
        data={0 0}
 
    case 40195
        data={65535 0}
 
    case 50000
        data={0 0}
 
`

gavinying, 2025-01-07:
Do you have the SunSpec Modbus protocol? why not follow the document straight away? also interested to know if your goal is to read data from actual SunSpec device, or to create a SunSpec emulator?