top-maker/modpoll #82: Can't get modpoll to run in a Compose Setup

mircsicz, 2025-01-06:
First and foremost thanks a lot for this awesome tool and all the work you've put into creating and still put maintaining it! :-*

I'm setting up a proof of concept environment in Docker-Compose but always fail at running the command in the container:
```
Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: exec: "modpoll -d --tcp-port 5020 --tcp modsim --mqtt-host modmqtt --config /app/examples/modsim-docker.csv": stat modpoll -d --tcp-port 5020 --tcp modsim --mqtt-host modmqtt --config /app/examples/modsim-docker.csv: no such file or directory: unknown
```

TLDR;

This is my compose.yml:
```
---
services:
  modsim:
    image: topmaker/modsim
    container_name: modsim 
    restart: unless-stopped
    #environment:
    ports:
      - 5020:5020
    volumes:
      - ./modsim-config:/config
    networks:
      - modbus
  modpoll:
    image: topmaker/modpoll
    container_name: modpoll 
    restart: unless-stopped
    command: ["modpoll -d \
    --tcp-port 5020 \
    --tcp modsim \
    --mqtt-host modmqtt \
    --config /app/examples/modsim-docker.csv"]
    volumes:
      - ./modpoll-config:/app/examples
    networks:
      - modbus
  mosquitto:
    image: eclipse-mosquitto
    container_name: modmqtt
    restart: unless-stopped
    ports:
      - "1884:1883"
      - "9002:9001"
    volumes:
      - ./mqtt-config:/mosquitto/config
      - ./mqtt-data:/mosquitto/data
    networks:
      - modbus
networks:
  modbus:
```

I've read through your Readme and also gone over this [Writeup](https://levelup.gitconnected.com/read-modbus-data-with-python-modpoll-tool-06e2128777ad). The Doc's I've seen as well...

If I comment out the whole 'modpoll' service from the compose file and run it directly from 'docker run' it works:
```
 0:43:32 CODE/docker/modpoll 
$ docker run --rm -v /Users/mirco/CODE/SNB/docker/modpoll/modpoll-config:/app/examples --name modpoll --network=modpoll_modbus topmaker/modpoll \
  modpoll -d \
--tcp-port 5020 --tcp modsim --mqtt-host modmqtt \
--config /app/examples/modsim-docker.csv

Modpoll v1.4.1 - A New Command-line Tool for Modbus and MQTT

2025-01-06 23:43:40,105 | I | modpoll.main | Setup MQTT connection to modmqtt:1883
2025-01-06 23:43:40,119 | I | modpoll.main | Connected to MQTT broker.
2025-01-06 23:43:40,119 | I | modpoll.modbus_task | Loading config from: /app/examples/modsim-docker.csv
2025-01-06 23:43:40,120 | I | modpoll.modbus_task | Added 1 device(s)...
2025-01-06 23:43:40,120 | I | modpoll.main | Loaded 1 Modbus config(s).
2025-01-06 23:43:40,120 | I | modpoll.main |  === Modpoll is polling at rate:10.0s, actual:10.0s ===
2025-01-06 23:43:40,120 | I | modpoll.mqtt_task | Created new MQTT session.
2025-01-06 23:43:40,121 | I | modpoll.mqtt_task | Subscribe to topic: modpoll/+/set with QoS: 0
2025-01-06 23:43:40,121 | I | modpoll.mqtt_task | Subscribed successfully.
2025-01-06 23:43:42,134 | I | modpoll.mqtt_task | Publish message to topic: modpoll/modsim01/data
```
I can even check with MQTT-Explorer:
[image]

gavinying, 2025-01-07:
Hi @mircsicz, thanks for your kind words. 
Your compose file is almost correct, the only issue is the format of `command` in modpoll service, you could give a quick try by putting the whole command in one line (removing the backslash, brackets and double quotes). 
Furthermore, you could use below format to make it looks better,
```
    command: >
      modpoll -d
        --tcp-port 5020
        --tcp modsim
        --mqtt-host mosquitto
        --config /app/examples/modsim-docker.csv
```
Here is my complete test code, FYI.
```
---
services:
  modsim:
    image: topmaker/modsim
    container_name: modsim 
    restart: unless-stopped
    #environment:
    ports:
      - 5020:5020
    # volumes:  # <-- Not needed for this example
    #   - ./modsim-config:/config  # <-- Not needed for this example
    networks:
      - modbus
  modpoll:
    image: topmaker/modpoll
    container_name: modpoll 
    restart: unless-stopped
    command: >
      modpoll -d
        --tcp-port 5020
        --tcp modsim
        --mqtt-host mosquitto
        --config /app/examples/modsim-docker.csv
    volumes:
      - ./modpoll-config:/app/examples
    networks:
      - modbus
  mosquitto:
    image: topmaker/mosquitto
    container_name: modmqtt
    restart: unless-stopped
    volumes:
      - mosquitto-config-vol:/mosquitto/config
      - mosquitto-data-vol:/mosquitto/data
    networks:
      - modbus
    ports:
      - 1883:1883
    environment:
      - MOSQUITTO_USERNAME=test
      - MOSQUITTO_PASSWORD=test
networks:
  modbus:
volumes:
  mosquitto-config-vol:
  mosquitto-data-vol:
```
> I used my own mosquitto image here, so that I don't need to mount extra config file. You are free to change to official one with your current config. 

Also, I think this is a great idea to put `modpoll` and `modsim` together for a docker compose example, I'll probably add it in our examples to benefit more developers. Welcome your contribution if you are comfortable to pull request your final working examples :)

mircsicz, 2025-01-07:
Thanks for your reply, I knew it couldn't be that much...

I"ll craft a PR to your Docu later, but please be so kind give me a hint: I imagine you want that in your Readme, or do you want it in examples/ as addition to 1-... or as a new 3...?

gavinying, 2025-01-07:
I think `examples/3-xxx` will do, name it like "3-test-with-local-modsim" or you may suggest a better name for it. Thanks in advance.

gavinying, 2025-01-07:
Just a reminder in case you need, this repo follows [conventional commits](https://www.conventionalcommits.org/en/v1.0.0/), please follow the convention when you draft commit messages, so that the CI pipeline won't complain about it. For this case, prefix `docs:` shall be a good fit. \
Let me know if you have any doubt.

mircsicz, 2025-01-16:
I honestly have never sent a PR here on GH, so far only on Enterprise porjects where I had all the permissions...

Been going through an Digital-Ocean [Article](https://www.digitalocean.com/community/tutorials/how-to-create-a-pull-request-on-github) but can't figure how to... :-(

I come as far as:
```
git push --set-upstream origin new-branch
```

gavinying, 2025-01-24:
Have you successfully [forked the repository](https://www.digitalocean.com/community/tutorials/how-to-create-a-pull-request-on-github#fork-the-repository)? 
Once you forked the repository, it will show up in your github account and you will have all permissions (because it is your repository after forking). 
Remember to clone your own (forked) repository to your local PC and add my repository as "upstream" origin, just keep in mind, all rest steps are working on your own repository.