top-maker/modpoll #84: Is it possible to map input registers to a single bit?

MattCosturos, 2025-02-14:
Many devices use bit-packing to merge 16 alarm signals to a single register.
Is it possible to poll a resister, but make a reference to a single bit value?

Example using :1 for MSB and :16 for LSB
```
poll,input_register,30000,1,BE_BE
ref,Level_Sensor_High_High_Alarm,30000:1,bool,r,
ref,Level_Sensor_High_Alarm,30000:2,bool,r,
ref,Level_Sensor_Low_Alarm,30000:3,bool,r,
ref,Level_Sensor_Low_Low_Alarm,30000:4,bool,r,
ref,Level_Sensor_Malfunction_Alarm,30000:16,bool,r,
```

Example using :15 for MSB and :0 for LSB

```
poll,input_register,30000,20,BE_BE
ref,Level_Sensor_High_High_Alarm,30000:15,bool,r,
ref,Level_Sensor_High_Alarm,30000:14,bool,r,
ref,Level_Sensor_Low_Alarm,30000:13,bool,r,
ref,Level_Sensor_Low_Low_Alarm,30000:12,bool,r,
ref,Level_Sensor_Malfunction_Alarm,30000:00,bool,r,
```

gavinying, 2025-03-13:
@MattCosturos, sorry for the late response. 
Yes, it might be a good way to handle bit values. Thanks for your suggestion. I'll look into the solution when I am free. Currently still busy with other stuffs.

gavinying, 2025-12-06:
Implemented in v1.6.0.